cornellhyperloop/electrical #315: Create FSM test plan - documentation and code

ridhitbhura, 2023-02-20:
https://github.com/cornellhyperloop/electrical/blob/master/State_Estimation/src/statemachine.cpp

ridhitbhura, 2023-02-20:
https://drive.google.com/drive/u/0/folders/1Km5-EVelO-qi-dWL58uK6_jURQFDM--z

ridhitbhura, 2023-03-20:
Creat test tutorial on Confluence @davidsw34 @Aimobhani9

agh93, 2023-09-25:
Planning
[https://confluence.cornell.edu/display/culoop/FSM+Test+Cases](url)

agh93, 2023-10-02:
Setup Google Test:
https://confluence.cornell.edu/display/culoop/How+to+use+Google+Test

4b causing error. First errors:
C:/Users/aghec/Documents/Cornell/hyperloop/electrical/State_Estimation/src/statemachine.cpp:239:3: error: 'WriteData' was not declared in this scope
  239 |   WriteData((char *)"Open", 4);
      |   ^~~~~~~~~
C:/Users/aghec/Documents/Cornell/hyperloop/electrical/State_Estimation/src/statemachine.cpp: In function 'void closeBrakes()':
C:/Users/aghec/Documents/Cornell/hyperloop/electrical/State_Estimation/src/statemachine.cpp:258:3: error: 'WriteData' was not declared in this scope
  258 |   WriteData((char *)"Close", 5);
      |   ^~~~~~~~~
C:/Users/aghec/Documents/Cornell/hyperloop/electrical/State_Estimation/src/statemachine.cpp:260:3: error: 'ReadData' was not declared in this scope; did you mean 'readData'?
  260 |   ReadData();
      |   ^~~~~~~~
      |   readData
C:/Users/aghec/Documents/Cornell/hyperloop/electrical/State_Estimation/src/statemachine.cpp: In function 'states accelerate(double, double)':
C:/Users/aghec/Documents/Cornell/hyperloop/electrical/State_Estimation/src/statemachine.cpp:277:6: error: expected '(' before 'checkDistance'
  277 |   if checkDistance (traveledDist, totalDist - decelRange)
      |      ^~~~~~~~~~~~~
      |      (
C:/Users/aghec/Documents/Cornell/hyperloop/electrical/State_Estimation/src/statemachine.cpp:281:3: error: 'else' without a previous 'if'
  281 |   else
      |   ^~~~

closed 2024-02-12 by davelil4